acm-cfw/acm-cfw.github.io #1: `sunxi-fel` Linux binary not included in acm-install_20221201.zip package

farmerbb, 2022-12-02:
When running the `acm-backup-and-install.sh` script on Linux, the installation fails because the `sunxi-fel` binary is not found in the data folder.

Install script output:

```
ACM batocera installation
=========================

1. Copy the folder install to a USB key
2. Insert the USB key into the 2P port of the ACM
3. Connect the usb power cable to the ACM power port
4. Press a key when you are ready, and then plug the USB cable to this computer.

*** PRESS A KEY WHEN READY AND THEN CONNECT THE ACM TO THE USB ***
No Allwinner devices in FEL mode detected.
Attempt #:1
No Allwinner devices in FEL mode detected.
Attempt #:2
No Allwinner devices in FEL mode detected.
Attempt #:3
No Allwinner devices in FEL mode detected.
Attempt #:4
Initializing...

Sending now...

finished: status:0...31

Astro City Mini should be in FEL mode now.

Booting from memory...
./acm-backup-and-install.sh: line 20: sunxi-fel: command not found
./acm-backup-and-install.sh: line 21: sunxi-fel: command not found
./acm-backup-and-install.sh: line 24: sunxi-fel: command not found
./acm-backup-and-install.sh: line 27: sunxi-fel: command not found
./acm-backup-and-install.sh: line 30: sunxi-fel: command not found
./acm-backup-and-install.sh: line 36: sunxi-fel: command not found
Done. The installation will continue on the ACM. Follow the ACM screen prompts.
The ACM will reboot when the hack is installed.
```

Data folder listing:
```
fes1.bin          libwinpthread-1.dll  sunxi-autofel-mac        sunxi-fel-mac
install_uRamdisk  restore_uRamdisk     sunxi-fel.exe            u-boot-sun8iw5p1_memboot.bin
libusb-1.0.dll    sunxi-autofel        sunxi-fel-handshake.exe  uImage
```

acmeplus, 2022-12-02:
Sorry, I forgot to mention that you should install sunxi-fel from your package tool on Linux. It’s typically provided as part of sunxi-tools. I’ll look into adding it to the package.

farmerbb, 2022-12-02:
Just came to the same conclusion myself :slightly_smiling_face: 

`sudo apt install sunxi-tools` works on Ubuntu to get `sunxi-fel` installed.  Installing the hack as we speak...

farmerbb, 2022-12-02:
Batocera is up and running on my Astro City Mini.  Appreciate your hard work!

closed 2022-12-05 by acmeplus