acm-cfw/acm-cfw.github.io #2: Dependencies to use acm-backup-and-install.sh

benoitb85, 2022-12-02:
I tried it on my old Debian buster (32bit computer) and got a "Exec format error". So I then tried on my home server that is running Debian 64 bits, latest stable version.

As mentioned in issue 1, I first installed this Debian package: sunxi-tools

Then I tried the acm script. 

First got this error:
./data/sunxi-autofel: error while loading shared libraries: libfdt.so.1: cannot open shared object file: No such file or directory

I installed these debian packages:
  libfdt-dev libfdt1

Now I'm getting these errors:
./data/sunxi-autofel: /lib/x86_64-linux-gnu/libc.so.6: version `GLIBC_2.33' not found (required by ./data/sunxi-autofel)
./data/sunxi-autofel: /lib/x86_64-linux-gnu/libc.so.6: version `GLIBC_2.34' not found (required by ./data/sunxi-autofel)

Is there a recommended linux version to use ? I did not have more success on Windows with Zadig or the cllassic driver.

benoitb85, 2022-12-02:
Got it working by installing Ubuntu 22.10 on my computer and the sunxi-tools package. Installation is ongoing.

acmeplus, 2022-12-02:
Thanks for testing that. I’m using Ubuntu 22.10 64bit. I suspect the gcc version on your Debian version is different. In any case I’ll upload the latest sunxi-autofel repo later so anyone can recompile if needed. 

Let me know how the installation goes. It should be uneventful on Linux.

zx81zx81, 2022-12-03:
I can confirm that the installation went fine on my ACM using Ubuntu 22.04.1 64bits (i had first to install sunxi-tools).

benoitb85, 2022-12-03:
> Thanks for testing that. I’m using Ubuntu 22.10 64bit. I suspect the gcc version on your Debian version is different. In any case I’ll upload the latest sunxi-autofel repo later so anyone can recompile if needed.
> 
> Let me know how the installation goes. It should be uneventful on Linux.

I first had a problem where the installation wouldn't complete. I reflashed the batocera img to another usb drive and it then went perfectly fine. Now I could play a bit with Batocera. I copied a neogeo game that I own and it seems to show up but I don't have the right bios in the right place it seems. I'll educate myself about Batocera tomorrow.

The installation did work, and thank you so much for developing this and releasing it. I've been dreaming of using my Astro for some NeoGeo and CPS games.

acmeplus, 2022-12-03:
> I first had a problem where the installation wouldn't complete. I reflashed the batocera img to another usb drive and it then went perfectly fine. Now I could play a bit with Batocera. I copied a [...]

Very nice. Batocera has great resources online, just visit the main [wiki](https://wiki.batocera.org) or their discord channel if you have any questions. For neogeo yes, you will need to have the necessary bios and put it under /userdata/bios/.

acmeplus, 2022-12-03:
For those of you who want to recompile autofel for your linux distribution, the branch is [here](https://github.com/acm-cfw/sunxi-tools/tree/autofel).

closed 2022-12-11 by acmeplus