acm-cfw/acm-cfw.github.io #4: No Wif-Fi Networks Found

loganknecht, 2022-12-05:
Hello!

# Gratitude

First, as always. I am so thrilled for the work you have done. Thank you so much! The fact that I'm even booting `Batocera` on this thing is AMAZING.

# Batocera Specs

| Type | Spec |
| - | - |
| Astro City Mini Version | Alpha 0.1 |
| Batocera Version | batocera-sunxi_r16-acm-36-20221201.img |

# Problem

I have gotten to boot into `Batocera`'s default menu, and managed to remap the controls.

I am now trying to get the `WiFi` network to work.

If I 
1. Go to the `Main Menu` 
2. Go to `Network Settings`
3. Select `Enable WiFi`
4. Select `WiFi SSID`

I don't see my network, or any networks listed.

If I select the `Refresh` button, nothing happens.

I tried restarting the device and that didn't change anything.

I have also tried inputting the SSID/Password in manually and that didn't work either.

# Questions

1. Does the `Sega Astro City Mini` have a wifi chip/card?
2. Does the `Sega Astro City Mini`'s wifi chip/card work on `2.4 Ghz` or `5Ghz`?
3. Is there anything special that needs to be done to get on WiFi?
4. Is there any speculation that you can do on my situation for why I can't connect to my network?

acmeplus, 2022-12-05:
First question is, do you have a usb wifi dongle? The ACM does not have either wireless or bluetooth internally, you need to use a usb wifi dongle based on a realtek 8188us or 8182cu chipset. For instance [this one](https://www.amazon.com/gp/product/B008IFXQFU/ref=ppx_yo_dt_b_search_asin_title?ie=UTF8&th=1).

drapos, 2022-12-05:
Hi Acmeplus,

Thanks for your efforts!

Any chance of increasing the chipset support for WiFi/ethernet dongles?

I have a couple of unused USBs there that would be sweet to be able to use here:
WiFi (RTL8192EU TL-WN823N) / Ethernet (RTL8153 TP-Link UE300).

Thanks for your time!

loganknecht, 2022-12-16:
Closing as `Not Implemented`

@acmeplus Is there a location or a `README.md` that I should be placing this information in for others so they can find the answers? More than happy to open merge requests.

closed 2022-12-16 by loganknecht

acmeplus, 2022-12-16:
> Closing as `Not Implemented`
> 
> @acmeplus Is there a location or a `README.md` that I should be placing this information in for others so they can find the answers? More than happy to open merge requests.

The documentation is [here](https://acm-cfw.github.io), and its repo [here](https://github.com/acm-cfw/acm-cfw.github.io), so you can fork and create a merge request with changes if you want to update it or create/extend a FAQ section.