acm-cfw/acm-cfw.github.io #5: Can The Astro Mini USB Connections Be Expanded?

loganknecht, 2022-12-05:
Hello!

# Gratitude

First, as always. I am so thrilled for the work you have done. Thank you so much! The fact that I'm even booting `Batocera` on this thing is AMAZING.

# Batocera Specs

| Type | Spec |
| - | - |
| Astro City Mini Version | Alpha 0.1 |
| Batocera Version | batocera-sunxi_r16-acm-36-20221201.img |

# Problem

I have my `Player 2` port using the usb mounted flash drive of `Batocera`

I have the `Player 1` plugged into an `Astro City Mini Arcade Stick` and I have the default controls on the device itself.

If I wanted to expand this to 4-players, is this possible? Do we know the answer to this yet?

Additionally would I be able to plug additional other devices as well, like say a network card?

acmeplus, 2022-12-05:
Yes, you can use a usb hub on either port. How many depends on the type, hub, etc. But usb hubs should work. Now I don't know if anyone has tested actual 4P but there's no reason they shouldn't work if the emulator supports it.

loganknecht, 2023-02-14:
Hello!

Thank you again @acmeplus !

I have tested with a very basic usb hub and it worked brilliantly.

I'm closing as `solved`

closed 2023-02-14 by loganknecht