acm-cfw/acm-cfw.github.io #6: Vsync issues

drapos, 2022-12-05:
Hi!

First of all, thanks for sharing this work with everyone. I'm having a great time testing this release.

I'm reaching you because I'm noticing problems with VSync (Horizontally since the screen is apparently rotated).
This is more aparent during flashes or with certain systems like the Game Gear.

Again, many thanks for making this happen!

acmeplus, 2022-12-05:
There's little that can be done regarding that. The gpu driver (mali 400MP) is closed source and there's little support for it. The screen is indeed rotated, and most of the internal applications have been patched to get them rotated. 

If we eventually manage to get this ported to a newer kernel, then there's the possibility of using the lima open source driver, but the amount of effort to get that done goes beyond this current project. Maybe someone will take some interest to help in that area.

lucke, 2023-01-08:
@drapos @acmeplus  just to throw some info on this topic, I have a dedicated custom rpi4 set up in vertical mode. I found that Batocera 35 and 36 have severe VSync issues in both retroarch and ES. However, batocera v32 works perfectly with a dedicated vertical set up without VSync issues. So it might also be some kind of Batocera issue. Thought it would be helpful to share. Thanks for your work!!

lucke, 2023-01-08:
@drapos @acmeplus having said that, I've found some retroarch settings that completelly remove the VSync issues and tearing. I've tested it with FBNeo and horizontal games and makes them great! 

Using retroarch this is the setting:
- Settings -> Video -> Syncrhonization:
- Turn VSync OFF
- Turn Sync FPS to content to ON

To make it permanent, add the following lines into batocera.conf

```
global.retroarch.video_vsync = "false"
global.retroarch.vrr_runloop_enable = "true"
```

drapos, 2023-01-10:
Editing batocera.conf worked wonderfully.

Thanks for sharing!

RygorX, 2023-01-17:
**YES !** ive been banging my head against the retroarch config to improve this ....thanks so much, this made a huge difference not 100% now but much much better . Metalslug background scrolling was so choppy prior to this now its just noticable.

RygorX, 2023-01-17:
hmm also frame rate displayed for vertical shooters Armored Batrider and Battle Gregga have increased from 40fps to 59-60