acm-cfw/acm-cfw.github.io #7: Which `MAME` Rom Set Version Should Be Used With Batocera

loganknecht, 2022-12-06:
# Gratitude

As always, thank you for everything!

# Batocera Specs

| Type | Spec |
| - | - |
| Astro City Mini Version | Alpha 0.1 |
| Batocera Version | batocera-sunxi_r16-acm-36-20221201.img |

# Question

I have found the latest roms here on Archive.org for `0.250` of the `MAME` roms
https://archive.org/details/mame-merged

But when I try to load these with `Batocera` in my `Sega Astro City Mini` it's incredibly hit or miss.

I have tried placing these roms in the `root` `roms/mame` directory or the `roms/mame/mame2003` directory, but neither seems to help with compatibility.

- Does anyone know what the appropriate `MAME` roms set to use with this version of `Batocera` is?
- Does anyone know what to do to improve compatibility?
- Do I need to update the `MAME` bios?

acmeplus, 2022-12-06:
ACM does not support the latest mame. The best mame versions supported in the ACM are 2000 (0.37), 2003 (0.78), and 2010 (0.139). 2000 and 2003/2003+ are probably the ones with better performance for the ACM.

lucke, 2023-01-06:
@acmeplus is your comment also applicable for the E2M regarding the mame versions? Thanks!!

fakemaria, 2023-01-06:
> @acmeplus is your comment also applicable for the E2M regarding the mame versions? Thanks!!

Is the same

loganknecht, 2023-06-06:
Closing as answered.

closed 2023-06-06 by loganknecht