acm-cfw/acm-cfw.github.io #8: v0.2 Scaling options not working anymore

drapos, 2022-12-07:
Hi!

I have just dropped into the v0.1 USB the updated files, and found this issue.

Batocera started with screen rotated 90º anticlockwise.

I corrected that manually, but now no scaling option works anymore.

I was using Core defined scaling+Integer scaling to get the best scaling option in my opinion, but now no setting there makes any difference neither on built in screen or hdmi external output.

HDMI output is working fine if you plug out and in after the system is already booted up.

Thanks for your efforts!

drapos, 2022-12-09:
Something more to add:

Dreamcast.

Flycast (the one without Libreto) is rendering 90º anticlockwise. If you enable rotate screen 90 degrees, it adds another 90º anticlockwise, so it renders upsidedown instead.

This one is better than Libreto: Flycast since it performs better, doesn't crash and allows you to select Vulkan to gain some extra FPS.

Surprisingly, MvC2 is almost there. Playable but with frame drops.

Cheers!

loganknecht, 2022-12-10:
Hello 👋 

I have been experiencing some inconsistencies with rotation and `HDMI` input as well.

--------------------
When I start with an `HDMI` input connected I will have to unplug it and re-plug it back in.

--------------------
By default in the `Alpha 0.2` version the screen is rotated counter-clockwise 90 degress, so I have to set it to be rotated 90 degrees clockwise normal.

However, when I connect the `HDMI` input it shows the screen now rotated incorrectly by that 90 degrees clockwise setting. So then I have to go back and undo the rotation setting for the `HDMI` input.
When I do that it show

--------------------

I'm not sure what to do about this other than fiddle with the settings and disconnect/reconnect the `HDMI` input after starting up.

acmeplus, 2022-12-11:
At this point unplug/plug the HDMI is the best solution. Just try to avoid doing it too quickly because the board may not detect it on time and get confused.

There is already an issue to set the default orientation automatically during the first boot, but won't be fixed until the main repo is synchronized again with batocera upstream. So while that happens (don't expected it soon) just use the ES menu to set the orientation, or use the unplug/plug method.

acmeplus, 2022-12-11:
For the scaling there may be a regression in the retroarch patches that I'll need to review. There may be some inconsistent rotation code in the shaders management code in retroarch. That may explain why v0.1 had it working scaling/shaders and 0.2 seems to fail.

loganknecht, 2023-06-06:
Hello 👋 

I'm following up on this.
I believe I'm incredibly close to being able to use the `ACM` as an emulation device.

This appears to be my last issue.

If I start the device it will load up correctly. Then if I connect an HDMI cable the system menus will look correct. However when I start the `Outrun` ROM in `FBNeo` the game starts `half screen` with the contents shown on the left and the screen is also rotated 90 degrees.

I can use the `retro arch` menu to rotate it to the correct size, but then it's stuck half-width and I'm not sure what to do to fix this. It's been a bit of a headache.

Do you have any suggestions or configurations on what to do for this situation?