acm-cfw/acm-cfw.github.io #10: Sega Astro City Arcade Stick In Player 1 Port Is Same Input For Player 2

loganknecht, 2022-12-10:
# Gratitude

I absolutely LOVE being able to play different MAME roms. This is my first time being able to do so and I love that I've been able to use the aesthetic of this device and its accessories to enjoy this nostalgia trip.

Thank you so much!

# Batocera Specs

| Type | Spec |
| - | - |
| Astro City Mini Version | Alpha 0.2 |
| Batocera Version | batocera-sunxi_r16-acm-36-20221206.img |

# Problem

This may be confusion on my end, but when I plug in my `Sega Astro City Arcade Stick` to the `P1` port on the back of the device. I set the inputs to map to `Player 2` in the configuration menu.

At this point my expectation is that this input has been configured AS `Player 2` which is separate from `Player 1`.

However, when I start a game, for example `Metal Slug 4` and I press `Select` to insert token and `Start` to play - I then arrive at the `Player Select` menu and it has both players active.

Then when I move with the `Sega Astro City Arcade Stick` ONLY the `Player 2` cursor moves, and the same goes for selection.

If I move the stick on the `Sega Astro City Mini` that controls the inputs for `Player 1`, and the same goes for selection.

# Questions

- Is there something wonky with the conifguration, or am I doing something wrong?
- Additionally, how do I map the buttons for `return to menu`, `insert token`, and `start` correctly
    -  In the controller mapping menu I only see a `Start` and `Select` option

loganknecht, 2022-12-10:
Ok - I have experimented with this a bit.

I have to go into the controller settings menu and then set each control manually.

The issue is that if I set the controller input for all the buttons like this

| Button | Mapping |
| - | - |
| Up Stick | Up D-Pad |
| Right Stick | Right D-Pad |
| Down Stick | Down D-Pad |
| Left Stick | Left D-Pad |
| Top Left Button | Y Button |
| Top Middle Button | X Button |
| Top Right Button | L Trigger |
| Bottom Left Button | B Button |
| Bottom Middle Button | A Button |
| Bottom Right Button | R Trigger |
| Credit Button | Select Button |
| Start Button | Start Button |

I then can't return to the main menu.

Does `Batocera` support multi-button combos for input?
How can I return to the system menu from in-game if all my inputs are mapped?

neocvera, 2022-12-11:
As you know the ACM itself only has 2 buttons for select and start, typically and is the case with the configuration recommended, hotkey is also mapped to select.

loganknecht, 2022-12-12:
Sounds good. Sorry for the noise.

Closing as cannot reproduce.

closed 2022-12-12 by loganknecht