acm-cfw/acm-cfw.github.io #13: `Batocera` `Quick Menu` Does Not Appear When `HotKey + Down` Is Pressed

loganknecht, 2022-12-15:
# Problem

When I am playing a rom I cannot bring up the `Batocera` quick menu

I am using two controllers

Controller 1 - Sega Astro City Mini
Controller 2 - Sega Astro City Mini Arcade Stick

I have configured the controllers with these settings

| Button | Mapping |
| - | - |
| Up Stick | Up D-Pad |
| Right Stick | Right D-Pad |
| Down Stick | Down D-Pad |
| Left Stick | Left D-Pad |
| Top Left Button | Y Button |
| Top Middle Button | X Button |
| Top Right Button | L Trigger |
| Bottom Left Button | B Button |
| Bottom Middle Button | A Button |
| Bottom Right Button | R Trigger |
| Credit Button | Select Button |
| Start Button | Start Button |

For the base `Sega Astro City Mini` controls I have set the `Select` button is set to the `Hot Key`.

I have read here that the `Hot Key` + the `B` button should bring up the quick menu during play, but nothing does.

https://wiki.batocera.org/remapping_controls_per_emulator

Instead, when I press the `Hot Key` button it's bringing up the arcade cabinet's `Main Menu` where you can test the controls and inputs.

And the reason I want to bring up the `Quick Menu` is because I need to remap the controls because the default mapping I have doesn't work. A single button is being mapped to multiple machine buttons when I play a game. For example `Attack` and `Special` buttons in the `X-Men` game are both being pressed when only the `A Button` is pressed.

# Questions

- Am I doing something wrong?
- Does the `Quick Menu` work for others?

fakemaria, 2023-01-03:
> # Problem
> When I am playing a rom I cannot bring up the `Batocera` quick menu
>
> I am using two controllers
>
> Controller 1 - Sega Astro City Mini Controller 2 - Sega Astro City Mini Arcade Stick
>
> I [...]

try adding these lines at the end of your batocera.conf file, which is located in share partition under system folder

global.retroarch.input_player1_a = "b"
global.retroarch.input_player1_a_btn = "5"
global.retroarch.input_player1_b = "a"
global.retroarch.input_player1_b_btn = "1"
global.retroarch.input_player1_l = "d"
global.retroarch.input_player1_l_btn = "2"
global.retroarch.input_player1_r = "c"
global.retroarch.input_player1_r_btn = "4"
global.retroarch.input_player1_x = "e"
global.retroarch.input_player1_x_btn = "0"
global.retroarch.input_player1_y = "f"
global.retroarch.input_player1_y_btn = "3"
global.retroarch.input_save_state = "x"
global.retroarch.input_save_state_btn = "nul"

loganknecht, 2023-01-03:
@fakemaria 

Thank you very much for the response.

I tried your suggestion and that did not have any effect.
I made sure the changes were applied to the `batocera.conf` file located in the `SHARE` partition.

What happened with these changes was the same as mentioned previously. When the `Hotkey + B` buttons are pressed the arcade machine's debug menu is shown.
It does not show the `Batocera` system menu.
Any combination of the `Hotkey` button with other buttons does not show the batocera system menu.

loganknecht, 2023-06-06:
Closing this - I had to reflash my batocera image. Something went horribly wrong for my first few attempts.

This now works with my Astro City Mini - But there are the screen rotation issues which is expected.

closed 2023-06-06 by loganknecht