acm-cfw/acm-cfw.github.io #14: Any plans to support the Astro City Mini V?

ktg5, 2022-12-25:
Just asking since v0.2 doesn't mention anything about supporting the Mini V or not.

acmeplus, 2022-12-27:
Astro City Mini V is not currently supported because we don't have access to a real model to test. Theoretically the only difference wrt the ACM is the screen, and the screen seems to be the same one that the Egret-II mini has. However, since the system requires a modified kernel, it's a risk to distribute a firmware without testing it.

ktg5, 2022-12-27:
I'd be willing to test it if it's possible to just boot from USB (not having to edit the system NAND) since I do have access to an Mini V.

loganknecht, 2023-06-06:
@acmeplus - If I were to purchase one and send it to you would that help?

acmeplus, 2023-06-07:
> @acmeplus - If I were to purchase one and send it to you would that help?

It's mostly a matter of time, and yes, donating a unit may work, but frankly at this time it's a matter of finding the time to rework a patch for a particular model. The last time I looked at it it seemed like the ACM-V has the exact same display as the E2M, so it should be technically feasible to create an updated image (it's just a single partition in the internal firmware) that could have that update. I'll keep it in mind but can't promise any timeline.