acm-cfw/acm-cfw.github.io #18: Batocera config improvement for volume control - lines provided

lucke, 2022-12-30:
Hello!

This should not an issue but an "idea", however I failed in finding that option, apologies.

I added the following configuration on the batocera.conf file to enable the following:

- Hotkey (select) + UP = Volume Up
- Hotkey (select) + DOWN = Volume Down

**I find this extremely convenient since there is no volume up/down buttons on the ACM** and though it would be good to share since in your release page you have anyway a section to add on the batocera.conf for proper button mapping.

Now, by default Save state slot+ and save sate slot- are mapped to hokey+up, hotkey+down; So I moved that to hotkey+right (save state up) and hotkey+left (save state down).

Here the config need to achieve the change I mention:

```
global.retroarch.input_volume_down = "down"
global.retroarch.input_volume_down_axis = "nul"
global.retroarch.input_volume_down_btn = "nul"
global.retroarch.input_volume_down_mbtn = "nul"
global.retroarch.input_volume_up = "up"
global.retroarch.input_volume_up_axis = "nul"
global.retroarch.input_volume_up_btn = "nul"
global.retroarch.input_volume_up_mbtn = "nul"

global.retroarch.input_state_slot_increase = "right"
global.retroarch.input_state_slot_decrease = "left"
global.retroarch.input_state_slot_increase_axis = "nul"
global.retroarch.input_state_slot_increase_btn = "nul"
global.retroarch.input_state_slot_increase_mbtn = "nul"
global.retroarch.input_state_slot_decrease_axis = "nul"
global.retroarch.input_state_slot_decrease_btn = "nul"
global.retroarch.input_state_slot_decrease_mbtn = "nul"
```

Feel free to add it to the guide, discard it or do as you see fit.

Thank a lot for your work!!

acmeplus, 2022-12-30:
Thanks for the contribution. I tried that before but I didn't want to change the default batocera shortcuts. The reason is that those are in their documentation and guides so it's better to use different combos for those actions. Maybe hotkey+L/R buttons.

lucke, 2022-12-30:
True, the only ones free are L/R (https://wiki.batocera.org/basic_commands)

Alternative Left/Right is used for rewind but I don't think we will be using rewind anytime for the ACM since it kills the performance

RygorX, 2023-01-18:
This works well, thankyou for sharing !