acm-cfw/acm-cfw.github.io #19: ACM Power button to shut down (Suggestion / Working)

lucke, 2022-12-30:
Hello,

Another improvement/suggestion with the actual fix.

Based on v0.2 version. I added the following file to allow shutdown by using the power button on the ACM.

Just create the file "multimedia_keys.conf" on /userdata/configs/

And add this line into the file:

```
KEY_POWER       0               shutdown -h now
```

And with that when moving the power swith to the right (off) it will shut down the machine.

Thanks a lot for your amazing work!

zx81zx81, 2023-01-02:
You can also add that line in the /etc/triggerhappy/triggers.d/multimedia_keys.conf file (you need to unsquash/mksquash batocera image)

acmeplus, 2023-01-02:
Thanks for the contribution. There’s a bug that prevents the proper batocera-shutdown to work. In the meantime that solution works. I’ll either fix it properly or add this workaround for the next release.

lucke, 2023-01-02:
@acmeplus FYI when doing the shutdown -f now batocera is doing a proper shutdown, because it shows the batocera logo and messages like "saving metadata etc", so I think batocera is listening to the OS event and shutting down properly. Try it, it's very acceptable.

acmeplus, 2023-01-02:
Yeah, I'm not concerned about your fix, I know it's working properly, but batocera keeps changing and I always try to keep it aligned with the baseline. In any case if I cannot solve the current bug I'll add the shutdown to the thd config.

Thanks

lucke, 2023-01-02:
thanks a lot for your work @acmeplus ! By the way, I just received an Egret 2 Mini and did the whole thing again. Works equally perfect. I updated the stock firmware of the E2M to the 1.1.0-171 version and all works. Thanks again!

fakemaria, 2023-01-02:
> thanks a lot for your work @acmeplus ! By the way, I just received an Egret 2 Mini and did the whole thing again. Works equally perfect. I updated the stock firmware of the E2M to the 1.1.0-171 version and all works. Thanks again!

you updated with our cfw installed or from a restored/ new unit?

lucke, 2023-01-02:
@fakemaria @acmeplus I did the following:

- Update E2M to latest stock version (from 1.0.2 to 1.1.0) via the instructions in their website (https://www.taito.co.jp/egret2mini/support/update)
- Install custom bootloader (yours) (https://github.com/acm-cfw/acm-cfw.github.io/releases/tag/e2m-alpha-v0.1)

All works.

I took a gamble there, I am aware. You can update the wiki if you want saying that your bootloader instalation is tested successfully in the latest stock firmware.

I was not willing to to the reverse (first install CFW, then do the official Firmware update) as I don't like to gamble that much :D Thanks!

fakemaria, 2023-01-02:
> @fakemaria @acmeplus I did the following:
>
> * Update E2M to latest stock version (from 1.0.2 to 1.1.0) via the instructions in their website (https://www.taito.co.jp/egret2mini/support/update)
> * [...]

thanks mate, that was I was asking. I had evidences that update isn´t working after installing the cfw. but the rollback script is working as expected. thanks for your time! thanks for the volume controls, that saves my day :)

marcosgeorgopoulos, 2023-01-05:
Hi, should this work on EM2 as well? I've tried it and I couldn't get it to work.

The one thing I will add is I had to create the 'configs' folder in /userdata as it didn't exist? Is the path you mentioned correct?

thank you

lucke, 2023-01-05:
@marcosgeorgopoulos hello! I'm sorry, I put the path wrong! (editing the OP now).

it's /userdata/system/config/multimedia_keys.conf

I've just tested it on the e2m and it also works there!

make sure the file has 755 permissions (rwxr-xr-x) (chmod 755 filename)

let me know how it goes

fakemaria, 2023-01-05:
> @marcosgeorgopoulos hello! I'm sorry, I put the path wrong! (editing the OP now).
> 
> it's /userdata/system/config/multimedia_keys.conf
> 
> I've just tested it on the e2m and it also works there!
> 
> make sure the file has 755 permissions (rwxr-xr-x) (chmod 755 filename)
> 
> let me know how it goes

fixed a typo in the path :)
/userdata/system/configs/multimedia_keys.conf

many thanks for the workaroud!

fakemaria, 2023-01-16:
**that script (volume or shutdown) introduces a regression on the dynamic rotation of the screen**

LukeUsher, 2023-01-17:
> **that script (volume or shutdown) introduces a regression on the batocera dynamic rotation of the screen**

The following contents will keep screen rotation working as expected
```
KEY_POWER       0               shutdown -h now
KEY_HOME+KEY_F 1                  batocera-audio setSystemVolume +5
KEY_HOME+KEY_C 1                  batocera-audio setSystemVolume -5
KEY_PLAYCD        1               /usr/bin/rotate_screen.sh 0
KEY_PAUSECD       1	          /usr/bin/rotate_screen.sh 1
```

fakemaria, 2023-01-17:
> The following contents will keep screen rotation working as expected
>
> ```
> KEY_POWER       0               shutdown -h now
> KEY_HOME+KEY_F 1                  batocera-audio setSystemVolume +5 [...]

But why introduce a rotation if that’s already inplemented?

LukeUsher, 2023-01-17:
> But why introduce a rotation if that’s already inplemented?

Rotation is implemented by having it stored in the default multimedia_keys.conf within the batocera root image.

When you create your own file in system/config, it *replaces* the original, rather than appends to it.

lucke, 2023-01-17:
Thanks, that explains why I accidentally broke the screen rotation. So the user’s multimedia_keys.conf file is overwritting the existing system multimedia_keys.conf file. That’s a bug because the batocera documentation sais that both can co-exist: https://wiki.batocera.org/scripting_function_keys“ Your custom multimedia_keys.conf file's configuration will be overlaid on top of the default profile. If there are conflicting settings, the one in /userdata/system/configs/multimedia_keys.confwill take priority.”