acm-cfw/acm-cfw.github.io #20: Missing Scrappers (only ArcadeDB on the list)

lucke, 2022-12-30:
Hello,

This time I don't have a solution (yet), using v0.2 when going to Batocera settings -> Scrape the only available scrapper is "ArcadeDB". It's missing the other 2 default ones included in emulation station (screenscrapper and thegamesdb).

Unfortunately ArcadeDB only has arcade games in the DB, so no other systems.

I'm trying to find how I can re-enable those 2 scrappers which are supposed to be part of ES but so far no luck, so this time I open the issue

Thanks!

acmeplus, 2022-12-30:
Full scraper support won't be available until the system is merged into batocera main. Only officially distributed batocera images have access to the different scrapers.

lucke, 2022-12-30:
Thanks @acmeplus. I'll close the issue and try with a command line scrapper

closed 2022-12-30 by lucke