acm-cfw/acm-cfw.github.io #21: black screen after e2m-backup-and-install.bat

yyoossk, 2022-12-31:
e2m-backup-and-install.bat sucsess
but console still black out

fakemaria, 2022-12-31:
> e2m-backup-and-install.bat sucsess
> but console still black out

Open the cmd with admin privileges and launch the bat file. Check where the error is. 
if the error is because it can’t find /data/ exe file then as a workaround do the following:
Copy /data content into the root folder (where bat file is)
Modify the .bat and remove the /data from the script
Run again

fakemaria, 2023-01-01:
> thx!

There is a new windows installer,please download and try

yyoossk, 2023-01-01:
e2m-backup-and-install.bat　error
Operation will stop.

yyoossk, 2023-01-01:
same not work buckup
black screen

yyoossk, 2023-01-02:
egret2-install_20221231.zip
same not work buckup
black screen
windows10

fakemaria, 2023-01-02:
@yyoossk you should add more information. That can be a zadig driver problems,or you are not using a proper usb a usb a cable,or you are not powering egret properly.
Remember that you can post screenshots that can help troubleshooting your issue.

Windows installation has been successfully installed on several machines so is a problem in your end, but hard to determine with so little info

Manch0i, 2023-01-04:
Hello, let me tell you in case it happens to anyone else, I changed the path of the SD roms to any external one and I always got a window with the message (we can't find any systems check that your paths are correct... .....) and there was no way to get out of there and finally I was able to solve it thanks to fakemaria's advice by formatting the SD and remounting the image batocera-sunxi_r16-e2m-36-20221107.img

Greetings.