acm-cfw/acm-cfw.github.io #23: retroarch menu orientation is wrong.

marcosgeorgopoulos, 2023-01-02:
Hi,

When I load up the retroarch menu in game, the orientation is the opposite to what it should be. 

When playing a game like street fighter in its correct screen orientation and I hit the hotkey + B, the menu is sideways,

adjusting the screen so its then vertical makes the menu the correct way up but doesn't display properly. 

I have added in the following configuration as per the documentation

`global.retroarch.input_rotate_landscape_btn = "14"
global.retroarch.input_rotate_portrait_btn = "15”`

acmeplus, 2023-01-02:
Thanks, yes, it needs to be fixed but it wasn't ready for the alpha. Will be fixed later on a beta release.

Note that if you manually modify the display.rotate you are going to break the auto rotation capability. So while that works to temporarily show the menu correctly it has several side effects.

neocvera, 2023-01-05:
Current fix: pop the screen out and rotate while you work through the menu (while the screen is popped out) - then return screen to its orientation, get back in the game.

marcosgeorgopoulos, 2023-01-06:
Thanks, however when I do this the menu is upside down.

fokhei, 2023-01-08:
I just change the menu driver to "rgui" on retroarchcustom.cfg, and it works perfectly!