acm-cfw/acm-cfw.github.io #25: Spinner not worknig

kaganita, 2023-01-04:
Hi,

I can't get the spinner to work.
When I plug it into the console it is detected.
I tested it in arkanoid, even if I choose it in the controllers menu, it is not working.
Interestingly, the trackball works in arkanoid, although very slowly

fakemaria, 2023-01-04:
Try that

In game, enter in retroarch options. You will be on quick menu. Go one step back to main menu, settings, input, port 2 controls, device index set to egret 2 mini paddle. 
Go back to quick menu, overrides, save game overrides

karlusp, 2023-01-04:
> Try that
> 
> In game, enter in retroarch options. You will be on quick menu. Go one step back to main menu, settings, input, port 2 controls, device index set to egret 2 mini paddle. Go back, mouse index, select the only available. Go back to quick menu, overrides, save game overrides

Thanks for your help.  I tried what you suggest and it doesn't work.  I tested other options and it doesn't work.  The only thing working is the trackball.  However, inside the retroarch menu, the spinner works to navigate menus

neocvera, 2023-01-05:
Test the latest update and please report back.

karlusp, 2023-01-06:
It still doesn't work in version 0.3 and even after testing fakemaria's suggestion.
I tested it on a version I upgraded from 0.2 to 0.3 and tested it on a clean 0.3 installation
The curious thing is that within the retroarch menu, the spinner works.

karlusp, 2023-01-22:
It's finally working

Quick Menu
"Back" to Main Menu
Settings
Input
Port 1 Controls
Mouse Index, change to Egret-II Mini Paddle

Even after I save, every time I reset, it's not loading that change for me
I try "Save controller profile" in port 1 controlls and in quick menu, Overrides, Save Games Overrides
From de Main Menu, i did a Quit, Restart System to force "saving metadatas"

Another problem, the spinner is spinning in reverse. Right to left and left to right

fakemaria, 2023-02-06:
> It's finally working
>
> Quick Menu "Back" to Main Menu Settings Input Port 1 Controls Mouse Index, change to Egret-II Mini Paddle
>
> Even after I save, every time I reset, it's not loading that change [...]

About spinner behaviour,you can change it in mame options