acm-cfw/acm-cfw.github.io #27: E2M Enhacements on Volume management and screen rotation on ES

lucke, 2023-01-05:
Hello,

Again, thanks a lot for your amazing work it is truly appreciated.

I've added 2 scripts to the E2M and captured them with events in order to:

1) Be able to rotate Emulation Station with the screen rotation. This basically listens to the screen, changes batocera.conf and reloads ES, it works. I have disabled it during retroarch execution (otherwise it was a mess and the hotkey is better for that). The only "drowback" is that if you rotate into retroarch, when going back to ES it is not rotated. This is mitigated by extracting the screen and pushing it in again without rotating it, or rotating the screen in ES instead of retroarch before launching the game.

2) Use the home button (which is the white button down from select/start) to rotate the volume by increments of 25% (0-25-50-75-100). I find this extremely convenient and doesn't tamper with any default batocera stuff.

Step 1: Listen to the events

edit the keys file
```
 /userdata/system/configs/multimedia_keys.conf
```
add this:

```
KEY_PLAYCD      1       /userdata/system/configs/rotateES.sh 1
KEY_PAUSECD     1       /userdata/system/configs/rotateES.sh 0
KEY_HOME        1       /userdata/system/configs/rotateVOLUME.sh
```

Now create the 2 scripts:

```
touch  /userdata/system/configs/rotateES.sh
chmod 755  /userdata/system/configs/rotateES.sh

touch  /userdata/system/configs/rotateVOLUME.sh
chmod 755  /userdata/system/configs/rotateVOLUME.sh
```
This is the content of the rotateES.sh script:

```
#!/bin/bash

# If retroarch is running PID is non null

PID=$(pgrep retroarch)
currentState=$(batocera-settings-get display.rotate)

# Checking if retroarch is running. If it is, we skip so that the hotkey rotates retroarch, but ES is not restarted

if [ -z "$PID" ];
then

# Retroarch is NOT running, therefore we can rotate ES
# Checking current screen position, if equal, don't rotate as it is not needed

        if [ -z "$1" ];
        then

                if [ $currentState != 0 ];
                then
                        batocera-settings-set display.rotate 0 && batocera-es-swissknife --restart
                else
                        batocera-settings-set display.rotate 1 && batocera-es-swissknife --restart
                fi

        else

                if [ $1 != $currentState ];
                then

                        batocera-settings-set display.rotate $1 && batocera-es-swissknife --restart
                fi
        fi

fi
```

This is the content of the rotateVOLUME.sh script:

```
#!/bin/bash

currentVolume=$(batocera-audio getSystemVolume)

if (($currentVolume == 100))
then
        currentVolume=-25
fi

newVolume=$(($currentVolume+25))
maxVolume=100

if (($newVolume > $maxVolume))
then
        batocera-audio setSystemVolume $maxVolume
else
        batocera-audio setSystemVolume $newVolume
fi
```

After doing everything, restart batocera so it can load the events in order to test it.

Hope it helps in improving the image!

fakemaria, 2023-01-05:
Thanks for your contribution !!!

acmeplus, 2023-01-06:
@lucke I can see the logic of the volume change, I would probably reduce the range to 0 to 75 or something like that, because 100% is a bit too much. 

However I don't know what you are trying to achieve with the rotateES and how that's different from what's already there. Both ES and RA already rotate, so not sure what's the difference here and if I'm missing something.

lucke, 2023-01-06:
@acmeplus for the rotate I was stupid. Basically I accidentally BROKE your implementation of rotation (because I rotated manually on first boot, contrary to your documentation) and I thought I was contributing by fixing it.

Long story short, only the volume part is interesting. Unfortunatelly I discovered that after booting a game and going back to ES the button doesn't trigger the volume change anymore... I'll keep working on it

Thanks!!