acm-cfw/acm-cfw.github.io #28: v0.3 video previews hanging up the UI

lucke, 2023-01-06:
Hello,

I've upgraded from v0.2 to v0.3

Preview videos now take like 1 minute to show when switching games, during this time the UI is hung up / sloppy.

I've tried with:
- 32 GB SD Card
- 128 GB SD Card
- FAT 32 roms parition
- exFat roms partition

And both on an upgraded installation and also a clean installation formating the SD card, result is always slow

I re-installed v0.2 and is lightning fast.

I guess this only happens when the library is big.

Can I provide some kind of log or can I look somewhere to give you more info on the issue? if so, let me know. I have both versions in 2 different SD cards, same roms library etc.

Thanks!

acmeplus, 2023-01-06:
Thanks for the report, I typically just test batocera default content, and those don't contain more than 1-2 videos per folder.

This update contained about 2500 changes on the batocera side of things so I'll have to check what's going on.

acmeplus, 2023-01-06:
@lucke I've just tried with a folder full of previews and I don't see any lag. One important question, did you update to v0.3 with the boot.tar.xz or installed it as a new sdcard from the img file?

ciaskohansen, 2023-01-06:
Hi, I had the same problem but after doing a clean install of the new image it hasn't happened again (for now).

lucke, 2023-01-06:
@acmeplus I tried both:

- First by using the boot.tar.xz to update from v0.2 to v0.3. When I faced the issue I thought "let's do a clean install"
- I did a clean install, formated SHARE to Fat32, copied my usual content and faced the issue.
- Then i formated to exFAT, copied my usual content and faced the issue.

If this is not happening to others, there might be something wrong with my conent, although it's working fine with v0.2

I'll try next time to just add roms and scrape them instead of importing pre-scraped content structure which probably contains some custom paths to the videos etc (my guess is that the issue is with custom paths to videos). Thanks!!

fakemaria, 2023-01-06:
Its reproducible either in update or clean install

Alexeyesgz0203, 2023-01-06:
I test， update to v0.3 with the boot.tar.xz or installed it as a new sdcard from the img file，both  video previews hanging up the UI

acmeplus, 2023-01-07:
I've been able to reproduce some of the behavior with a video backgrounds enabled for the systems view. It's not clear at this point if this is an issue with the latest batocera ES or the specific implementation for this board. 

As a workaround, disable video backgrounds, or revert back to v0.2

RygorX, 2023-01-18:
I get this hang issue , I think its reproducible withthe stock 0.3 install, selecting and unselecting the preinstalled space-twins gba game

paparansen, 2024-05-07:
no response here since ages.
donated 3$ only to write a message,
didnt even get an answer.