acm-cfw/acm-cfw.github.io #29: Fail to install on Windows 11

fokhei, 2023-01-07:
I am trying to install CFW alpha 0.1 to an E2M with win11 that already has the classic driver.

Every time I plug in the USB cable to connect the E2M to Windows, Windows can detect it (shows up as a Nintendo Classic or Allwinner FEL device, naming depends on the Classic driver), but always disconnects automatically the next second.

In this case, sunxi-fel-handshake init successfully and disappears, but fails to run in the next 5 steps and shows "ERROR: Allwinner USB FEL device not found!"

I tried this on 3 different Windows devices with 3 different USB cables and all are the same. Is there any way to help me?

Translated with www.DeepL.com/Translator (free version)

fakemaria, 2023-01-07:
> I am trying to install CFW alpha 0.1 to an E2M with win11 that already has the classic driver.
> 
> Every time I plug in the USB cable to connect the E2M to Windows, Windows can detect it (shows up as a "Nintendo Classic" or "Allwinner FEL device", naming depends on the Classic driver), but always disconnects automatically the next second.
> 
> In this case, sunxi-fel-handshake init successfully and disappears, but fails to run in the next 5 steps and shows "ERROR: Allwinner USB FEL device not found!"
> 
> I tried this on 3 different Windows devices with 3 different USB cables and 2 different classic drivers and all are the same. Is there any way to help me?

Do you have inserted the sdcard? Please post an screenshot of the screen error

davidarellano, 2023-01-07:
Hello, I have this same issue but trying to install the CFW in the Astro City Mini on Windows 10.

The failure appears in the same step described in [#1](https://github.com/acm-cfw/acm-cfw.github.io/issues/29#issue-1523926204)

I have checked that when plugging in the ACM to the PC using a data USB cable, the device appears in usbdeview using the correct driver, but it suddently disconnects. It only stays connected about 1 second. I get this same result after trying with several USB cables and also different USB ports of the computer, including USB 2.0, USB 3.0 and also using a USB hub with additional power.

After watching [a video (minute 3:23)](https://youtu.be/gAA94FXCNBk?t=203) I understand the script also involves some timing to get the device properly initialized, but although the device gets detected by sunxi-fel-hanshake some times quicker than others, it doesn't stay connected for any longer than that.

Here you have some screenshots, but I guess there is not much extra info about the issue:

[image]

[image]

davidarellano, 2023-01-07:
Yes, it is a desktop PC, both the cable and the wall are here 😁 .

I agree that it could be something related to the power of the USB port, but I already checked that the power settings are not interfering on that. I also checked on regedit that the device is not selective suspended or anything similar. 

Thanks for your response, I'll keep trying. I will also keep an eye on this conversation just in case somebody else finds a solution.

fakemaria, 2023-01-07:
> Yes, it is a desktop PC, both the cable and the wall are here 😁 .
>
> I agree that it could be something related to the power of the USB port, but I already checked that the power settings are not [...]

Only saw this error and was because an user was powering the egret with the usb from pc.
There was no other reports and the scripts were checked a d there are several success installations, so is a problem in your end.

fokhei, 2023-01-08:
> Do you have inserted the sdcard? Please post an screenshot of the screen error

yes, i have insert SDcard. Screenshot is 100%  same as https://github.com/acm-cfw/acm-cfw.github.io/issues/29#issuecomment-1374594842

fokhei, 2023-01-08:
Update: After many attempts (over 50), I finally managed to install it. But I still don't know what the reason for the success is.

My final solution was to keep the USB cable and power cable connected, then run e2m-backup-and-install.bat and press enter, then turn on the E2M.

If E2M boots to the official firmware, That means failure. Then you can power off the E2M and repeat this solution.

If successful, the E2M will remain black screen until it automatically reboots with a custom splash screen.

fakemaria, 2023-01-08:
> Update: After many attempts (over 50), I finally managed to install it. But I still don't know what the reason for the success is.
>
> My final solution was to keep the USB cable and power cable [...]

@acmeplus this sound more logical to be added to install guidelines

davidarellano, 2023-01-08:
Thank you for the update @fokhei!

Unfortunatelly that procedure didn't worked for me either, so I finally gave up on Windows. I booted my PC with a Ubuntu 22.10 live USB and it worked first try for both my ACM and E2M.

fokhei, 2023-01-08:
> Thank you for the update @fokhei!
> 
> Unfortunatelly that procedure didn't worked for me either, so I finally gave up on Windows. I booted my PC with a Ubuntu 22.10 live USB and it worked first try for both my ACM and E2M.

That is cool! Live USB Linux is a good way for Windows users also! Thanks for your tips!!!