acm-cfw/acm-cfw.github.io #30: ACM v0.2 - Vertical Arcade games rotated

lucke, 2023-01-07:
When using the v0.2 pre-release of the ACM if you run a vertical arcade game it appears rotated 90 degrees inside a vertical display on the 16:9 screen. It's very weird.

None of the normal retroarch rotation options when in the game fix the issue.

Tried with several vertical games with the same result. I believe this worked on v0.1 but can't confirm.

fakemaria, 2023-01-07:
> When using the v0.2 pre-release of the ACM if you run a vertical arcade game it appears rotated 90 degrees inside a vertical display on the 16:9 screen. It's very weird.
>
> None of the normal [...]

inside a system (eg mame) push select (list options),advanced options, and: vertical rotation (tate) to off and allow rotation set to off

lucke, 2023-01-07:
@fakemaria  thank you! Solved with these settings.

closed 2023-01-07 by lucke

fakemaria, 2023-01-07:
@acmeplus that settings is worth it to add it in a faq for acm