acm-cfw/acm-cfw.github.io #31: Abnormal shuttering?

PSPNDSGBASP, 2023-01-10:
Installed CFW Alpha 0.3 for my Erget Mini but all emulators I tried has shuttering every few seconds even the GB emulator...
Is it normal?
Thanks!

fakemaria, 2023-01-10:
Roll back to 0.1

PSPNDSGBASP, 2023-01-10:
> Roll back to 0.1

Thanks!
0.1 is perfectly fine for me!

firecrafts, 2023-01-15:
Im having same problem every few seconds it’s shuttering for 1 frame then back normal on mame/fbneo ( only tried arcade games and same result )

PSPNDSGBASP, 2023-01-16:
> Im having same problem every few seconds it’s shuttering for 1 frame then back normal on mame/fbneo ( only tried arcade games and same result )

After roll back to 0.1 everything fine.

Xam73, 2023-01-16:
One question, to roll back to 0.1 what I need? Only the batocera-sunxi_r16-e2m-36-20221107.img ? What is the exact procedure?
Thanks.

fakemaria, 2023-01-16:
if you want to reuse what you have, burn the 0.1 image in a new sd drive and extract/replace that file

[image]

Xam73, 2023-01-16:
Thanks for your reply! So I don't need to touch the Egret mini, only replace the file batocera in sd card /boot, right?

I want to roll back to 0.1 from 0.3 (shuttering problems)

fakemaria, 2023-01-16:
> Thanks for your reply! So I don't need to touch the Egret mini, only replace the file batocera in sd card /boot, right?
> 
> I want to roll back to 0.1 from 0.3 (shuttering problems)

nono, you don´t need to reflash!

Xam73, 2023-01-16:
Thanks fakemaria for your help!

0.1 works perfectly, no more shuttering