acm-cfw/acm-cfw.github.io #34: Network connection not working

Endprodukt, 2023-01-19:
Thanks for the great work, 

I can't get a network connection. I ordered 2 different ethernet to usb adapters and 2 different wifi dongles with the mentioned chipsets. Non of them work.

fakemaria, 2023-01-19:
Its an alpha and as an alpha this things are likely to happen. Plug and unplug until it gets connection

Endprodukt, 2023-01-19:
I know what's an alpha. If that's the argument there is no need to report any problems. I contribute to help. The mentioned chipsets didn't work on my end. That's all

fakemaria, 2023-01-19:
> I know what's an alpha. If that's the argument there is no need to report any problems. I contribute to help. The mentioned chipsets didn't work on my end. That's all

you aren’t reading it properly. Its an alpha,so there are problems like that in which you need to unplug and plug several times the dongle to make it work. 
That’s all.
You can read about that topic [here](https://github.com/acm-cfw/acm-cfw.github.io/issues/4)

fakemaria, 2023-01-19:
> The e2m documentation said to replug after booting and I did that many times but I'll keep trying. Thank you for the link.

I didn’t had too many problems with my units.
But with regards of tp link TL-WN725N first version isn’t working, only version 2 (I don’t have v3 so dunno)
Here is an attachment of v2
[image]

Znidz, 2023-04-17:
Old thread but...
I tried a no-brand USB Wi-Fi adapter and it didn't work even with unplugging/re-plugging rebooting etc.      
I bought a TP-LINK  TL-WN725N and it worked first time with no issues in port 1.

gsarris, 2023-04-18:
> > The e2m documentation said to replug after booting and I did that many times but I'll keep trying. Thank you for the link.
> 
> I didn’t had too many problems with my units. But with regards of tp link TL-WN725N first version isn’t working, only version 2 (I don’t have v3 so dunno) Here is a pic of v2 [image]

V3 works fine. Tested today on my unit 🙂

TheSupremacyCabinet, 2023-08-19:
Hello, I can confirm that this adapter for the Nintendo Switch also works very well with the TAITO Egret 2 mini
https://www.amazon.it/gp/product/B01N6S068R/