acm-cfw/acm-cfw.github.io #36: Flashing image to large SDCARD

putto72, 2023-02-20:
I found on Telegram an image (Melasopla) for Egret II Mini, which is for 32 Gb SDCARD. I suppose it is based on this image. I read somewhere that at boot if there is unused space it will be allocated as a new partition; is it correct? I was thinking about using a 512 Gb SDCARD to later add other roms... I have not tried patching firmware until now, I suppose that in this case it doesn't matter if I use your image or Melasopla, since what is patched is just the boot code which is not on the SDCARD, right? Thank you for your great work anyway!

putto72, 2023-02-22:
I answer my own question: I used Diskgenius to enlarge SHARE partition up to 512 Gb while flashing the image, then I patched my E2M (original firmware) having this card inside the unit. Everything worked perfectly, I will upgrade to version 0.3 to support trackball and spinner. Thank you!

closed 2023-02-22 by putto72

fakemaria, 2023-02-23:
> I answer my own question: I used Diskgenius to enlarge SHARE partition up to 512 Gb while flashing the image, then I patched my E2M (original firmware) having this card inside the unit. Everything worked perfectly, I will upgrade to version 0.3 to support trackball and spinner. Thank you!

0.1 image is compatible with trackball and spinner :)