acm-cfw/acm-cfw.github.io #38: Mame configuration (E2M)

putto72, 2023-02-26:
I'm using 0.3 image with Taito trackball/spinner connected to P2 USB. When running Marble Madness, to obtain the right responsiveness I must raise trackball sensitivity to 255 (both X and Y) by first enabling Mame Menu and opening Analog Settings. Unfortunately, I can't find a way to: 1) exit Mame menu, using just E2M buttons; 2) save my analog configuration. Any suggestions?
I'm also struggling with the freezing problems due to video playback on the menu; how can I disable videos (without deleting them)? Thank you!

fakemaria, 2023-02-26:
Use 0.1 instead,it has trackball and spinner support. Use the pad in port 1

putto72, 2023-02-27:
Thank you, I rolled back to 0.1; indeed trackball works in port 1 (just like in 0.3: I read incorrectly port number on the back). For Marble Madness I still have to increase sensibility to 255, though. I found that if I enable Mame menu, change analog settings, re-enter Quick Menu and disable Mame menu my analog preferences are preserved, so this is a viable solution. 
0.1 is more stable than 0.3; it does not freeze continuosly when browsing game lists, except once after booting the system and scrolling a long list - it takes about 30 seconds, then there are no more freezes. I can live with that! :)

closed 2023-02-27 by putto72

fakemaria, 2023-02-27:
> Thank you, I rolled back to 0.1; indeed trackball works in port 1 (just like in 0.3: I read incorrectly port number on the back). For Marble Madness I still have to increase sensibility to 255, [...]

You can try to reduce quality of images and videos,it will make the menus run smoother

Put also in your batocera.conf that line,to make both spinner and trackball usable

 global.retroarch.input_player1_mouse_index = 1