acm-cfw/acm-cfw.github.io #42: [E2M] MAME 0.78 plus doesn't seems to work with CPS1/2 and Neogeo romset for MAME 0.78 plus (2003 plus)

edd72, 2023-03-16:
When I use MAME 0.78 plus, CPS1/2 games and Neogeo games from Romset 0.78 plus don't launch.

Same for MAME 0.139 and its Romset for CPS1/2 and Neogeo.

Very strange.

Is there a bug for MAME core in general?

fakemaria, 2023-03-16:
For neogeo and cps use fba :)

TotalCoolness, 2023-03-17:
I think your neogeo bios is wrong. Did you look in the log file for errors ?

loganknecht, 2023-03-20:
@fakemaria What does `Use FBNeo` look like?

I would like to use it but I'm not sure what/how to configure it for my ACM. Can you elaborate a bit more on that? Do you have any links related to that you could share? Anything is appreciated.

drapos, 2023-04-03:
Just move your CPS1/2/3 & Neogeo roms to fbneo rom folder instead of MAME one, and they will show up under FBNeo emulator on ACM after refreshing games.

FBNeo romset might be required depending on the romset you have.