acm-cfw/acm-cfw.github.io #44: V-sync problem on the Last Batocera with Egret II mini

tomakorea, 2023-05-02:
on my version (I'm using the newest Batocera of the 2), there is a V-sync issue (but because the screen is turned by default, it's more like a h-sync). Anyway, as games on PC without V-sync tear the image, I have the same issue on Egret II Mini, just because the screen by default is rotated, instead of tearing the image horizontally, it does this vertically (line). Is there a way to fix this?

acmeplus, 2023-05-03:
Did you try the older version?

tomakorea, 2023-05-17:
Hi ! I just tried the old version, there is no tearing anymore but there is some refresh rate or frame pacing issues. I tried Super Mario Bros on the NES Emulator and when running at max speed the scrolling stutters and I doubt it's a performance issue. I also encountered the same issue on other emulators like FBNeo for example with UN Squadron.
Here are my settings in Batocera : 
VIDEO MODE : AUTO
VIDEO OUTPOUT : AUTO
GAME ASPECT RATIO : AUTO
GAME RENDERING SHADERS : NONE
BILINEAR FILTERING : ON
INTEGER SCALING : ON
VARIABLE REFRESH RATE : OFF
THREADED VIDEO : OFF
RUN AHEAD : OFF

As you see I tried to avoid every settings that is notorious to give stuttering or bad frame delivery, but I still got the same results.

It's not really a problem on games that have slow scrollings, but on fast horizontal scrolling games. Strangely it seems that this problem occurs randomly. 
I just tested, with the same settings on Super Mario Bros, First play Level 1 and 2, Start is smooth and after 5 to 10 second the scrolling start to stutter and never comeback smooth. Then I comeback to Batocera menu, I launch the game again, and now it's smooth for again 10 sec before starting to stutter. It looks like something happening during the game (memory leak?).

On the side note, I don't know if it's normal but the background music in the menus has micro drops also randomly (about every 10 to 15 seconds there is a super short drop with no sound). Maybe there is an issue with the sound driver that makes this a bit unstable? Could it be also related to the performance/scrolling stutter during emulation?
Let me know if you need a video of it, I can show you.