acm-cfw/acm-cfw.github.io #45: Fix for the V-sync and stuttering problem for Egret II Mini

tomakorea, 2023-05-19:
Hi, after noticing severe screen tearing and scrolling stuttering on emulators I found a fix to keep everything smooth. Unfortunately, theses options are OFF/FALSE by default and it's not possible to activate it in Batocera.. without a fix, it's necessary to bring the retroarch menu and activate the options manually everytime you launch a new game since the configuration file isn't saved.

The fix is to activate theses 2 options by default in retroarch : 
Configuration -> Video -> Synchronisation -> Hard GPU Sync = YES
Configuration -> Video -> Synchronisation -> Sync to Exact Content Framerate = YES

It seems like the screen in the Egret II mini is a variable refresh rate. Activating the 2 options fix not only the tearing, but also the frame pacing. When activating only Hard GPU Sync, there are still problems, so both options are mandatory.

To fix that : 

    Access the RetroArch Configuration File: You can find this file within the Batocera system. The location of this file will vary depending on your setup, but it's typically found at: /userdata/system/.config/retroarch/retroarch.cfg.

    Edit the RetroArch Configuration File: Use a text editor to open the retroarch.cfg file. You should see a lot of configuration options in this file. Search for the settings you want to change:

        For "Hard GPU Sync", look for a setting named video_hard_sync = "false" and change it to video_hard_sync = "true".

        For "Sync to Exact Content Framerate", look for a setting named video_sync = "false" and change it to video_sync = "true".

However, I'm on windows 10 and I couldn't mount the image (maybe it's also read only), so a fix would be greatly appreciated !

Thanks for your attention.

tomakorea, 2023-05-19:
My Bad !
Actually the config file is on the SHARE partition.
No need to bother the dev ! 

The settings to change in the retroarch config file : 
video_adaptive_vsync = "true"

However, I don't find the "sync to exact content framerate" option in the config file.. I'll investigate more

sorry for the confusion !

acmeplus, 2023-05-20:
No worries. Note that you can also override retroarch variables via batocera.conf. Just set global.retroarch.variable=value.

tomakorea, 2023-05-20:
Thanks :) I finally found everything !
Hard GPU Sync is mandatory and the only way is to edit the config file as I detailed above.
For Variable Refresh Rate it was hidden in batocera itself, just go to Main Menu -> Game Settings -> Latency Reduction -> Variable Refresh Rate : ON

It solves 99% of the tearing and scrolling issues. I noticed sometimes a slight stutter during diagonal scrolling, so maybe when the screen is moving horizontal + vertical it may stutter a bit, but honestly, the experience now is so much better than before, it's mostly fine.
Now I have to find a way to adjust the intensity and scale of the scanlines because the shader Zfast that does scanlines doesn't scale nicely with the 1024x768 screen of the Egret II.

skodkassen, 2023-06-28:
This hack is dead, isn't it? No more work will ever be done, right?

sonicteam77, 2024-05-04:
@tomakorea ，THANKS，I made the modifications as you said, and the picture synchronization is indeed improved a lot than before, but there is still stuttering problem when running the game in all emulators. The picture would glitch every few seconds, and the game audio is same problem