acm-cfw/acm-cfw.github.io #46: FBNeo RetroArch Menu Not Showing Selection Cursor

loganknecht, 2023-06-04:
Hello,

Thank you very much for making this possible!

I am still not able to fully realize this as an emulation solution.

---

# Batocera Specs

| Type | Spec |
| - | - |
| Astro City Mini Version | Alpha 0.2 |
| Batocera Version | batocera-sunxi_r16-acm-36-20221206.img |

# Problem

I cannot navigate the  in-game `RetroArch` menu (not sure if this is the right name for the menu).

It shows a black screen and my key presses work, but they're completely unknown as to what I've selected.

# Steps To Reproduce

1. Turn on ACM
2. Select `FBNeo`
3. Select any `FBNeo` `ROM` (Currently trying to get `Outrun` working
4. Wait for game to load
5. Press `HotKey` + `B`

Then the menu shows up? But it really doesn't.

It's completely black. Then I have to mash around random keys and if I'm lucky get to another menu. But I have no idea what is selected, and the keys are completely random for navigation. To be honest it's a nightmare.

See images below.
[image]
[image]

The end goal here is to be able to change the configuration of the controls so I can drive with a joystick instead of a steering wheel, but there's just no way for me to do any in-game configuration like this. It's impossible.

Am I doing something wrong? Is this a configuration issue? I feel like I'm going crazy.

fakemaria, 2023-06-04:
Is that hapenning without having connected to the hdmi? Sounds like they are assets missing on retroarch config folder

loganknecht, 2023-06-04:
Yes it happens without or without HDMI.

loganknecht, 2023-06-05:
Following up on this and closing.

I had reflash my batocera image. I have no idea why my previous installation was so borked, but it was. Re-flashing and testing it seems to be working.

Still need to test with HDMI and TV.

Thank you @fakemaria for the response. 👍