acm-cfw/acm-cfw.github.io #48: `FBNeo`/`RetroArch` On Left Half Of Screen And Rotated On Start

loganknecht, 2023-06-06:
Hello!

Hello 👋

Thank you so much for this custom firmware!

I believe I'm incredibly close to being able to use the ACM as an emulation device.
This appears to be my last issue.

If I start the device it will load up correctly. Then if I connect an HDMI cable the system menus will look correct. However when I start the `Outrun` ROM in `FBNeo` the game starts half screen with the contents shown on the left half and a black screen on the right half. Additionally the screen is also rotated 90 degrees.

I can use the retro arch menu to rotate it to the correctly, but then it's stuck half-width and it doesn't look correctly offset. I'm not sure what to do to fix this. It's been a bit of a headache.

Do you have any suggestions or configurations on what to do for this situation?

See images below
[image]
[image]

acmeplus, 2023-06-07:
@loganknecht quick question, what's is your ACM model, ACM or ACM-V? AFAIK the HDMI was never working because it required to undo the video rotation patches. In order to make it to work on either the ACM screen or HDMI display it requires some additional configuration work.

loganknecht, 2023-06-07:
@acmeplus I'm using the `ACM` model. I do have an `ACM-V` available as well though.

Are there any notes or documentation regarding the rotation pieces? Has anyone else encountered this as well?

acmeplus, 2023-06-07:
> @acmeplus I'm using the `ACM` model. I do have an `ACM-V` available as well though.
> 
> Are there any notes or documentation regarding the rotation pieces? Has anyone else encountered this as well?

Did you try rotating in retroarch video settings? As far as I remember getting EmulationStation rotated is just a configuration option, getting retroarch rotated may work but the video viewport will be cropped.

For ES rotation, there are variables in batocera.conf where you can set the system rotation. Experiment with those. For Retroarch let me know how switching the settings work.

For the ACMV is actually easier because I believe the display is not rotated, and in fact the vertical orientation is the native orientation. However the kernel will need to be updated, as I mentioned before.

loganknecht, 2023-06-07:
@acmeplus I have tried rotating it in retro arch. Unfortunately this is what it looks like.
[image]

I can confirm that Emulation Station works fine.

RE: ACM V
Are you saying I could use my ACM V successfully without issue? Or am I misunderstanding that.

acmeplus, 2023-06-07:
What you are seeing is because the actual video frame is rotated. Retroarch and ES can easily rotate, but the bezel is managed directly by the SDL2 code and does not to retroarch rotation.

About the ACMV, it's technical feasible but cannot be done without compiling the kernel and other elements again.

loganknecht, 2023-06-07:
@acmeplus I believe I understand what you are saying, but please confirm as I feel like I'm not quite grasping it.

It sounds like you're saying because of the SDL2 (Simple Direct Media Layer 2?) code there is nothing I can do to resolve this issue despite both Retro Arch and Emulation station also supporting rotation. And that no matter what I do to tweak either of them this will not be resolvable for the my HDMI output goals.

Is that correct?

Charlyfd, 2023-10-08:
Hello.

I have exactly the same problem. If I understand correctly, there is currently no way to fix this in the ACM.