acm-cfw/acm-cfw.github.io #53: Does CFW support latest firmware (v2.0.0-1) for Egret II mini?

jackiemoonsoon, 2025-04-10:
Hi there. I would like to install the latest firmware on the Egret II mini before installing CFW to use Batocera. Has anyone tested this on the latest Egret II mini firmware available (currently: v2.0.0-1, released on August 23, 2024)? 

https://www.taito.co.jp/en/egret2mini/support/update

jackiemoonsoon, 2025-04-22:
Is anybody still here to help with this question? :-/

Macs75, 2025-04-23:
Hi, from what on the telegram group yes it is possible to install the 2.0.0-1. If you are doing on an original firmware you can do it directly. If you are doing it on one that has already the CFW then you have first to restore the original firmware with the e2m-restore script.

Macs75, 2025-04-23:
please don't use github to ask questions and close this issue.